technobly/VEXTREME #14: Hardware STM32 SPI2 support for addressable LEDs

technobly, 2019-12-31:
## Feature

Currently the addressable LEDs are bit-banged with software SPI.  Ideally we'd like to support HW SPI2 so that we can do fancy things like DMA control of the LEDs.  To enable that, we'll likely have to shuffle some pins around.

Notes:
- Move V-HALT and V-CE to different pins and fix SDAT and SCLK so they are on SPI2 for hardware control. 
- V-CE might be optimized for MSB (seems like it is in romemu.S), so move to PA15 which bumps CSn to some other pin? 
- CSn should likely be on SPI1_SS, but it might not matter (check). PB01 is used in an optimized way as well though.

Alterac, 2020-03-03:
What about changing the part your using for RGB Leds to something simpler (and easy to source from lcsc) like the WS2813B and then you can use only 1 Pin, say PB8 - SPI5_MOSI to control them?

technobly, 2020-03-03:
Thanks for the idea @Alterac! WS2812B/13's are very timing dependent... so these would not be as flexible and resilient as Clock&Data IMO, which can be done very slowly if necessary.  We have tons of pins available... so it's just an exercise of juggling resources at this point.  WS2812B/13's are quite large as well.  I do agree on the sourcing issue though... there must be an easier/cheaper place to get these LEDs from than Digi-Key.

Alterac, 2020-03-03:
The PCB has lots of room, so i wouldnt be too worried about that, but 10 RGB leds is still quite extravigant imho, I think 4 would be more than enough (but whatever, i can just not solder on the rest, hah). But the point about clocking them is much easyier since you dont need to babysit the timing code.