technobly/VEXTREME #24: Add cartridge RAM support

technobly, 2019-12-31:
## Feature

Add cartridge RAM support (specifically for Animaction $2000-$27FF, but possibly for others as well). 

Any downside to letting the entire 32K/64KB of emulated ROM to be considered as RAM and writable?

phillipeaton, 2019-12-31:
I'd like the ability to have more RAM, 1kb is a bit small for Forth
development.

Cheers,
Phil

technobly, 2019-12-31:
@phillipeaton Animaction is 2Ki, but the desire would be to just enable the whole thing to be writable.  Do you see any downside to that?  How much RAM does Fourth require, and what memory address range do you currently use?  Do you have to do anything special in your program to enable it on VecFever?

phillipeaton, 2020-01-01:
The downside of it all being RAM is that some bad code might try and write
to the ROM address space, if it's ROM, it's a harmless write, if not...

Do any programs do that? Probably not, but who knows? I guess it would be
OK to make all the non-ROM address space work as RAM up to the 48k limit.

Forth (not Fourth) in the early days was bootstrapped from disc into RAM,
there was no ROM. Nowadays, RAM and ROM is plentiful and cross compilers
exist, so you can get by with less. Currently I'm using whatever is left in
the Vectrex 1k that isn't BIOS reserved, probably about 750 bytes, which is
good enough for some buffers, stacks, user variables and quite a few test
words, but you couldn't dump a whole program onto the Vectrex and let it
compile it into RAM. Running a Forth multi-tasker in 750 bytes is also
probably a non-starter, though not sure what the performance hit would be
anyway.

Thomas told me that extra RAM could be configured in the VecFever, but we
never got round to setting it up, so all I'm using that is special in the
VecFever is an extra serial port. Originally he supplied a Prolifics PL2303
device, but that has dodgy drivers on Windows, so he gave me an FT232
device that works much better. The serial port is enabled using some bytes
in the cart header block, the baud rate is setup in the VecFever menus. It
also provides 256 byte buffers, probably for send and receive.

I can send you the cart header block of you're interested.

technobly, 2020-01-01:
From Malban:
>About the RAM: I can tell how the VecFever does it :-). Per default the complete "ROM" area is writeable (AKA RAM). But there are some games that actually (bugy!) write to ROM and destroy the running game (some of pcJohn's games) - so that not always is a good option. So the VecFever allows to switch on a ROM mode, where the ROM area is not writeable! Binaries for the VecFever are saved in "V4E" format... that format includes a header for various options... (like size, bankswitch scheme, RAM/ROM needs cartridge options etc).

If the V4E header schema is open and sharable, please do share it.  I don't know why it wouldn't be open since it's for creators of games for VecFever, but I can't say that I've run across it anywhere :)  I don't want to copy it, but there could be value in parsing it so you wouldn't have to create separate ROMs for VecFever and VEXTREME.

I believe Malban uses the extra RAM in VecFever for Vectorblade.  48Ki of RAM.

gtoal, 2020-04-17:
A bit mask on the vextreme side could be used to mark RAM areas as writable or read-only.  Depending on the memory available in the vextreme to store this bitmask, a bit could modify read/write behaviour for anything between 4K blocks all the way down to single bytes.  Also if more than one bit is used, it could represent more than just a read/write flag - it could be used to trigger an NMI interrupt to be serviced by the 6809 which would allow the 6809 to implement virtual peripherals rather than requiring the virtual peripheral support to be baked in to the ARM side (eg as per Philip Eaton's request for an addressible serial port - with this a read could be trapped and the NMI code could call the vextreme via RPC before placing the fetched value in a register or at a fixed addess in 6809 memory like a DMA transfer)  This mechanism could also be used for placing breakpoints in the rom for debugging 6809 execution.

technobly, 2020-04-18:
Since VEXTREME now ignores all files besides .vec/.bin case insensitive, I was thinking we could just put a .txt file with the same name as the game that is loaded, where special settings could be enabled just for that game.  Most games would just use the default settings and not require this file.  We could make a special file type and/or extension, but I feel like that's just going to confuse people.  Everyone can edit a text file, right?  If it gets saved without the .txt extension it would still work.

E.g.
Animaction.bin
Animaction.txt
```
# VEXTREME_ROM_OPTIONS_FILE
#
# RAM_RW - RAM Read/Write
# <R/W OPTION>,<START>,<END>
# READ_ONLY,0,7FFF (default), 
# READ_WRITE,<START>,<END>
RAM_RW=READ_WRITE,2000,27FF
```

closed 2020-05-26 by technobly

technobly, 2021-05-11:
FYI: I ran into these things recently!

@phillipeaton 
> "The downside of it all being RAM is that some bad code might try and write
to the ROM address space, if it's ROM, it's a harmless write, if not...
Do any programs do that? Probably not, but who knows? I guess it would be
OK to make all the non-ROM address space work as RAM up to the 48k limit."

@malbanGit
> "But there are some games that actually (bugy!) write to ROM and destroy the running game (some of pcJohn's games)"

---

So I implemented the RAM cart specific options tonight :D

- I first modified the romemu.S to allow for variables to be used for RAM enable, start, and end address
- Created a Cart Options structure
- Created a Cart Options check/load/get/close function
- For every file loaded, it also checks if there is a matching *.ini file... if so it attempts to load it.  The options get parsed and the game is executed with RAM supported at the desired location, or not.  It works great!

Now RAM will be disabled unless we request it by adding a matching game.ini file.  The ini file is binary data, but super simple to create by hand, and rarely used.

```
40 00 01 00 01 20 7F 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

data is encoded in little endian format

0x0040 = 64 bytes for size
0x0001 = Version 1 Cart Options format
0x01 = bit 1 set, RAM enabled (other bits not used currently)
0x20 - high byte of start address (aka 0x2000)
0x27 - high byte of end address (aka 0x27FF)
This is for Animation of course

If you wanted to only allow 0x2000 to 0x20FF, you would specify it like this
0x20 - high byte of start address (aka 0x2000)
0x20 - high byte of end address (aka 0x20FF)

Logo Lite would be
0x60 - high byte of start address (aka 0x6000)
0x7F - high byte of end address (aka 0x7Fxx)

Technically Logo needs RAM all of the way to 7FFF, but we will run into a problem with that if it ever reaches the top, because that's how we call our RPC functions.  So, we'll need to move those to >=0x8000 ;-) But for now you'll be hard pressed to reach 0x7FFF with Logo Lite.

PR #61 (2020-05-26): Implements Animaction RAM Support
### Problem

Animaction was not working.

### Solution

Enables Animaction support by implementing the write operation back to cartData[] 0x2000 ~ 0x27ff (originally from @rattboi in PR #30 ).  More work had to be done to solve why this wasn't working though; it turned out to be 16-bit writes were not supported in the original code, so operations like STD $ADDR were only writing the first byte.  This is heavily used by Animaction.

### Steps to Test

- Compile for v0.3 (USE_HW=v0.3)
  - Obviously play Animaction!
  - Double check Polar Rescue still works
  - R [... cut]