technobly/VEXTREME #31: Better Reset Heuristic

technobly, 2020-01-03:
## Enhancement

### Currently the reset mechanism works like this:
- reset a counter when ROM emulation starts (menu is loaded of a game is requested to be loaded)
- increment a counter every time the cartridge is read from
- when you see the copyright symbol address 0x0004 being read, check if that counter is above a certain value
- if so, assume the game has been reset and the cart header is being read again and take action to bring the menu back up
- if not, just keep on reading and incrementing the counter

### New Reset Heuristic

- We need a better heuristic for determining when the Vectrex has been reset.
- Games like Polar Rescue have bugs that end up causing the copyright header addresses to be read during the game causing false tripping
- Also this current time based + address detection heuristic is not that reliable... for example if a game had a really long intro music tune it would cause false tripping (I don't actually understand why long intro music causes the reset timer to show as expired, because the header is read before the music starts playing, but that is the observed result)

### References

#22

gtoal, 2020-01-09:
Doesn't a press of the reset button always cause a fetch of the 6809 reset vector?  Would that be sufficient to detect it?

technobly, 2020-01-09:
Yes that’s something I’m currently exploring. Thanks @gtoal !

technobly, 2020-01-13:
Closed by #37

closed 2020-01-13 by technobly