technobly/VEXTREME #38: Y-Music crashes / plays incorrect tune

cparsonsuk, 2020-01-12:
## Bug Report
Everything except Monty on the Run and Robocop crash the Vextreme requiring reset.
The two songs that don't crash play the incorrect music and display the incorrect image.

### Expected Behaviour
All songs play the correct music and display the correct image when selected

### Observed Behaviour
Screen goes blank when most songs selected.
Monty and Robocop do not crash but play the incorrect tune

### Steps to Reproduce
Vextreme 0.2

### References

Links to a video, or more info

technobly, 2020-01-12:
Thanks for the report @cparsonsuk !  I can confirm this is also not fixed by the new reset mechanism #31

technobly, 2020-04-18:
I just realized YMusic.bin is 64KB and VEXTREME must not deal with that bank switching correctly.  Is the bank switching algorithm that YMusic.bin uses known?  @cparsonsuk I know you are the other Chris, but maybe you know?

technobly, 2020-04-18:
Karl_Quappe_64.bin runs on VEXTREME, and I'm pretty sure it's accessing the upper 32KB because the high score text MAL is shown, which lives in that bank.

cparsonsuk, 2020-04-20:
Hey Brett, I'm not sure which bank switching method Chris Malcolm uses, it
might be one other than the one provided via VIDE which is, presumably, the
way Malban does Karl Quappe.
Chris

commit ac210f25 (2025-08-10): fixes #38, 64kb bankswitched games work now
closed 2025-08-10 by technobly