technobly/VEXTREME #40: Ability to load and run code on STM rather than on 6809

gtoal, 2020-01-13:
Would like to be able to write code that runs on the STM32 and sends vector drawing commands to code running in RAM on the Vectrex's 6809, in order to have a mechanism to add new games, whether written from scratch in C or static binary translations of arcade games into C.  Requires a new format of file to be recognised, a 6809 bootstrap to move code into Ram, and some drawing support code on both sides.  It is not necessary to emulate memory while such a program is running, but there has to be a mechanism for resuming the memory emulation on exit from the STM-hosted code.

gtoal, 2025-07-08:
*ROFL*

phillipeaton, 2025-07-08:
This prompted me to look at the other open issues and especially the RPC serial port one I raised and you replied to in 2020, for which I've now taken a closer look at the code and now I understand what STM32 RPC is!