technobly/VEXTREME #42: Screen saver?

gtoal, 2020-01-22:
After leaving my vectrex powered up all day it occurred to me that the bright "VEXTREME" at the top of the screen might lead to burn in, in the long term.  Perhaps after a period of inactivity, the default screen could fade out everything except the word VEXTREME, and have it bounce across the screen in a path that eventually covers all positions.  Original screen to be restored on any joystick/button activity.

technobly, 2020-04-21:
That's a good idea @gtoal !  How about after a timeout, the menu turns into a mini game? It would just be in demo mode until you grabbed the controls, and then, you'd be in!  Button 1 would exit.

technobly, 2020-05-04:
BTW, I have built Developer Mode #23 from the ground up with screen wear in mind ;-)

PR #73 (2021-03-20): Revamped menu
### Problem

- Font is hard to read and ugly
- Navigation is cumbersome, no indication of where you at currently in the list
- LED is too bright and fleshy, there is no way to change brightness or completely turn it off
- No screen saver to protect CRT from burn-in

### Solution

- "Raster" based font replaced by vector font. It's drawn using optimized routines borrowed from Malban's source codes. Whole range of ASCII chars supported. Currently lower and upper case letters are the same, but the table made in such way that this can be changed easily in the future. Filename is crop [... cut]
closed 2021-03-20 by technobly