technobly/VEXTREME #50: Create a serial port in the Vectrex address space

phillipeaton, 2020-04-15:
A serial port connected to a terminal emulator on a PC is handy for debugging and allows interactive environments like forth of basic to run.

I'd like a serial port to run something similar to my serial port handler that works on a VecFever (code below is in Forth):

HEX

\ VecFever UART serial port interface. VecFever provides buffering (256 bytes each way?)
7FFB EQU v4eTxStatReg \ Read, negative if transmit buffer is in use, positive otherwise
7FFB EQU v4eTxByteReg \ Write
7FFC EQU v4eRxStatReg \ Read, zero if no data received, otherwise != 0
7FFD EQU v4eRxByteReg \ Read, upon reading, 7FFC is automatically cleared
7FFE EQU v4eLEDReg    \ LED Register, probably read/write?

CODE KEY?   \ -- f    return true if char waiting
  6 # ( D) PSHS,   CLRA,   v4eRxStatReg LDB,
  NE IF,   -1 # LDB,   THEN,   NEXT ;C

CODE KEY    \ -- c    get char from serial port
   6 # ( D) PSHS,   BEGIN,   v4eRxStatReg LDB,   NE UNTIL,
   v4eRxByteReg  LDB,   CLRA,   NEXT ;C

CODE EMIT   \ c --    output character to serial port
   BEGIN,   v4eTxStatReg  LDA,  MI UNTIL,
   v4eTxByteReg STB,   6 # ( D) PULS,   NEXT ;C

**Describe alternatives you've considered**
Other alternatives are available, but a VEXTREME is cheaper than the other alternatives and/or potentially provides a lot more functionally.

gtoal, 2020-04-17:
This - and any other virtual devices added by the vextreme - might be placed in the C000-C7FF space so as not to interfere with the rom space (0000-BFFF).  Although I'm not sure why a virtual device interface is better than the existing RPC mechanism for calls to the Vextreme?

phillipeaton, 2020-04-17:
All I know about RPC is that it can mean Remote Procedure Call. How would I
use that in place of a serial terminal? Can you point me to the code?

gtoal, 2020-04-17:
> All I know about RPC is that it can mean Remote Procedure Call. How would I use that in place of a serial terminal? Can you point me to the code?
> […](#)

here's the led test which sends commands to the vextreme arm by rpc - look for rpcfn near the end of the file:
https://github.com/technobly/vextreme/blob/master/code/led/led-test1/led-test1.asm

I don't know for sure that serial I/O is available via this mechanism but if it isn't, it should be!