technobly/VEXTREME #70: Files and/or folders won't show up in the menu

Badummzi, 2020-07-20:
Hey technobly!

I recently sent you this issue via your e-mail, but since I think that you don't check it that often, I'll repost it here 😉 I hope that's ok!

Anyways, I'm having the following issue (from my e-mail):

I got my hands on a vextreme a few days ago, and now I've ran into an issue, that I have no idea how to fix...
I built and installed the firmware correctly, set up the menu successfully, after I had formatted the drive in windows with the correct settings. Everything seems to work just fine and I am also able to copy files onto the device.
However, i can't get anything other than the menu to work. I tried a lot of times in different ways, but the directory "roms" or any other never shows up in the menu. If I then press button 4 on the controller, the vectrex just crashes and if i reset the device, it goes to a buggy version of mine storm. What am I doing wrong? Could it be a fault on the pcb?
 
P.S. I had the cart plugged in the wrong way at first and the vectrex didn't boot at all, but after i flipped it everything worked fine (I hope that didn't do anything!) I also tried reformatting and cleaning the drive, creating a new partition and even reflashed the firmware, but to no avail. I'll put a video of me showing this problem as an attachment to this. Btw, I should also say, that I'm rocking a European Vectrex.
 
I hope that is something, that is easy to fix!

Since I'm not ably to upload mp4s here, I'm uploading it inside a zip-file.

[issue.zip](https://github.com/technobly/VEXTREME/files/4949530/issue.zip)

Thanks in advance! 😄

plasticbrain777, 2020-07-20:
Have you tried the latest compiled release from https://github.com/technobly/VEXTREME/releases/tag/v0.23

Badummzi, 2020-07-21:
@plasticbrain777 Thank you for your answer! :D But after extensive troubleshooting, I just realized, that I hadn't installed the right driver for windows 🙄 Now it works! ;)

closed 2020-07-21 by Badummzi