technobly/VEXTREME #72: Be able to change LED-Brightness

Badummzi, 2020-10-26:
I'm currently using my Vextreme (btw, extremely happy with it!) without any casing and the only thing, which kind of annoys me, are the really bright leds facing upwards. 

It would be a cool feature, to be able to reduce the led brightness, or turn it off completely!

Alternatively have a DIP switch on the pcb or another firmware, to turn them off?

technobly, 2020-10-27:
They are pretty bright without a case, and without a label.  If you want to perform a hacky way to dim them, you can open up [LED Test v2 app](https://github.com/technobly/VEXTREME/tree/master/code/led/led-test2) and turn down the brightness, then exit that test app and they should stay set to the same brightness until you power off.  We will be implementing some options that will persist to adjust things in the future.

Badummzi, 2020-10-30:
Oh, thank you! :D Not an ideal way, but this will work for now! ;)

PR #73 (2021-03-20): Revamped menu
### Problem

- Font is hard to read and ugly
- Navigation is cumbersome, no indication of where you at currently in the list
- LED is too bright and fleshy, there is no way to change brightness or completely turn it off
- No screen saver to protect CRT from burn-in

### Solution

- "Raster" based font replaced by vector font. It's drawn using optimized routines borrowed from Malban's source codes. Whole range of ASCII chars supported. Currently lower and upper case letters are the same, but the table made in such way that this can be changed easily in the future. Filename is crop [... cut]
closed 2021-03-20 by technobly