technobly/VEXTREME #74: High Score Issues - Some games use the same title and save/load high scores under that same title.

LimitZer0, 2021-04-15:
There are some games with the same title (like All FURY UNLIMITED games). These games all save/load under the same game name ("F U R Y" in this case).

There may be more games. Different versions of the same game could have this issue as well (alphas/betas/final release/etc) if anyone cares.

technobly, 2021-04-15:
Yeah this is a known issue but I'm glad you opened a issue for it.  It bugged me that MINE STORM I and MINE STORM II only write to "MINE".  We need to figure out a way to hash the header and use that, or find the longer string amongst the header data.  I think we thought you could probably read up to a certain number of bytes (80? 255?) and terminate at that number or the end of the cart header which is indicated with 0x00 IIRC.

technobly, 2025-08-11:
This is fixed in v1.0